openmrn/OpenMRNLite #1: mDNS failing. And what port is Openmrnlite listening to?

dsj782, 2019-10-17:
First, what TCP port is OpenMRNlite listening to? I can ping the ESP32, but not sure which port I should configure in JMRI. (using the WiFi option).

Second, code uploads, starts, but then I get this:
Updating CDI file /spiffs/cdi.xml (len 6089)
[WiFiMgr] Starting WiFi Manager task
[WiFi] Starting TCP/IP stack
[WiFi] Initializing WiFi stack
[WiFi] Configuring WiFi stack
[WiFi] Starting WiFi stack
[WiFi] [1/36] Waiting for SSID connection.
[WiFi] Setting ESP32 hostname to "esp32_50101011823".
[WiFi] Starting DHCP services.
[WiFi] Station started, attempting to connect to SSID: MOTO8F3C.
Allocating new alias 8E5 for node 050101011823
[WiFi] Connected to SSID: MOTO8F3C
[DHCP] [2/36] Waiting for IP address assignment.
[WiFi] IP address is 192.168.99.43, starting hub (if enabled) and uplink.
[mDNS] Initializing mDNS system
[mDNS] Setting mDNS hostname to "esp32_50101011823"
[Uplink] Starting mDNS searching for _openlcb-can._tcp.
[mDNS] No matches found for service: _openlcb-can._tcp.
[Uplink] mDNS search failed.
[Uplink] Starting mDNS searching for _openlcb-can._tcp.
[mDNS] No matches found for service: _openlcb-can._tcp.
[Uplink] mDNS search failed.
[Uplink] Starting mDNS searching for _openlcb-can._tcp.
[mDNS] No matches found for service: _openlcb-can._tcp.
[Uplink] mDNS search failed.
[Uplink] Starting mDNS searching for _openlcb-can._tcp.
[mDNS] No matches found for service: _openlcb-can._tcp.
[Uplink] mDNS search failed.
[Uplink] Starting mDNS searching for _openlcb-can._tcp.
[mDNS] No matches found for service: _openlcb-can._tcp.
[Uplink] mDNS search failed.
[Uplink] Starting mDNS searching for _openlcb-can._tcp.
[mDNS] No matches found for service: _openlcb-can._tcp.
[Uplink] mDNS search failed.
[Uplink] Starting mDNS searching for _openlcb-can._tcp.
[mDNS] No matches found for service: _openlcb-can._tcp.
[Uplink] mDNS search failed.
Over and over.

Any ideas what is going on?

atanisoft, 2019-10-17:
The default behavior for the ESP32 is not to enable the HUB mode in the CDI. This is due primarily to the ESP32's performance on TCP/IP as a HUB leads to lost CAN frames and timeouts. You CAN enable it via the JMRI CDI editor *BUT* it will very likely have performance issues.

The mDNS behavior you are seeing is expected as the ESP32 is searching for a HUB that advertises itself as providing "_openlcb-can._tcp". Since it is not finding anything advertising that service it retries the search after a few seconds.

dsj782, 2019-10-18:
Thanks for your reply.I am totally confused on what a lcc hub is and how to configure it in JMRI. I can find nothing on the web on the hows, whats, and whys of a lcc hub.When I start the hub, it only shows the home address 127.0.1.1 (I thought home was 127.0.0.1?) and a port 12021 .I understand performance will be an issue, but if I keep each ESP32 limited to what it does, I should be ok. (maybe? lol)

The intent of the wifi esp32: 
I belong to the Detroit Model Rail Road club in Holly MI. We have a very large layout and there are areas where that are hard to wire.My main project is testing LCC. We currently have CMRI and it looks like both will be used on our layout.I would like to have the ESP32 talking to JMRI vi a network interface/WiFi.

As a side project, I am testing MQTT. I can convert CMRI to MQTT and MQTT to CMRI, using an USB attached ESP32, with MadLeech's library.
I would like to network connect the ESP32, running some sort of CMRI code, that can convert to MQTT.
Maybe do the same with LCC to MQTT/MQTT to LCC?
I have tried to do both on a raspberry PI, with no luck. But the ESP32 would be the preferred method.

Don James.

atanisoft, 2019-10-18:
> We currently have CMRI and it looks like both will be used on our layout.I would like to have the ESP32 talking to JMRI vi a network interface/WiFi.

Often times groups will start with a mixture of LCC and non-LCC components and over time migrate exclusively over to one system. As for JMRI talking to the LCC side via WiFi, that would be easiest using a rPi as the LCC hub. This can be done using the OpenMRN [hub](https://github.com/bakerstu/openmrn/tree/master/applications/hub) application. JMRI will talk to the hub and the ESP32 nodes should be able to find it via mDNS and connect.

> I can convert CMRI to MQTT and MQTT to CMRI, using an USB attached ESP32, with MadLeech's library.

Depending on the type of CMRI IO nodes you are using, you might be able to directly swap in an ESP32 as an LCC IO device to bridge the two systems.

> Maybe do the same with LCC to MQTT/MQTT to LCC?

I'm not sure there is much benefit in doing this or if it is even feasible.

atanisoft, 2019-10-19:
I've replicated the mDNS publish failure on the latest ESP-IDF/Arduino-ESP32 and have filed https://github.com/espressif/esp-idf/issues/4217. I'll be looking at options to improve this in the default configurations.

balazsracz, 2019-10-20:
I think there is a true missing feature here, namely that it is difficult to connect to a device that has been freshly flashed. Once the user had a chance to edit the configuration, there are a number of features that exist and useful:
- the hub can be turned on which allows a remote JMRI to connect to the esp32
- the uplink can be enabled with a fixed IP and port number, connecting to another hub that does not have mDNS (including a builtin hub in JMRI).

However, all of these are only achievable if the user can edit the configuration. That is only possible if the user managed to connect a JMRI to the esp32 network. So this is a chicken-and-egg problem.

I see a few ways to get out of this:
1) enable the hub listener to be on by default (we should not be advertising it via mdns as default though -- tat should be a setting with default off).
2) use the serial port via gridconnect and have the option to enable that (from code). Then JMRI can be connected as if it was a USB-CAN adapter and the config can be set that way.
3) add functions to connect to remote hub or have a way in the code to set the default fallback connection to the remote hub.

atanisoft, 2019-10-20:
I agree, there is a missing piece here. I'm not certain that enabling the hub mode is a good idea as the ESP32 has a hard time with serving the CDI when operating as a hub. From my testing it has less than a 50% success rate and when it doesn't load the SSID connection goes down and can not recover without a reset (bug in WiFi driver, Espressif already working on a fix).

Of the three options you have proposed the best would likely be the third. A fourth option to consider would be port my StateFlow httpd/dns code over to OpenMRN (uses BSD sockets so it is portable) and expose a version of the node config UI shown in JMRI but only for "this" node.

balazsracz, 2019-10-20:
I don't understand your fourth proposal, could you expand on it a bit more?

atanisoft, 2019-10-20:
JMRI node config dialog rendered as HTML directly on the ESP32, similar to this:
[image]
But rendered in a dynamic fashion based on the CDI.xml.

dsj782, 2019-10-23:
Mike,Sorry about the delay. I was on a much needed vacation. Now just catching up on emails.
Things I have learned this past few days:1. What mDNS is. Now I understand it (mostly). As a Firewall Engineer, I know DNS, but never heard of mDNS. I have heard of Bonjuor and did not know they were related.
2. What a hub is in LCC. And how it works in the standard client/server schema. 
3. The differences between an ESP8266 and ESP32. The ESP32 has a built in can controller, the esp8226 does not.4. IBM wrote MQTT and Node-Red.
 
I was able to get the OpenMRN working on the esp32!. 
With JMRI 4.16, there was a lot of issues with the CDL and the openmrn. I could only get the cdl open twice out of a lot of try's.But when I moved to 4.17.4, it works every time. Mostly.The openmrn's cdl only works when the LCC interface was set to "simulate lcc". Anything else, the configure cdl  tree only showed the node address, with no "open twisty thing". But the node still works.
Things I want to learn:How to setup the CDL screens.How to add more i/o ports.
I have done C++ in the past, but there are a few lines I have no idea what they are doing."    openmrn.stack()->print_all_packets();" looks strange to me. What and where is "print_all_packets()? I have never seen that kind of call before.
I still want to add some code that translates it to MQTT, when it receives a packet. And some code to send a packet when it receives a MQTT.Any idea where in the code I should add my code? I tried a few places, but no luck.
 

Working for one of the largest retailer back in the 80's and 90's, we used IBM's MQ a lot. Credit card swipes, price scans, inventory scanners, ........  Looks like MQTT is just a simpler version of MQ. 
So I have been task with looking at it from a model rail road prospective . While I like LCC, it does have a few drawbacks. And JMRI's implementation of MQTT is just starting. 
I use the "pubsubclient" library for  MQTT..

And thanks to you and your team for OpenMRN. This will help push LCC/OpenLCB in a better direction.

Our clubs is currently using JMRI with Dr Chubb's C/MRI. I am in the middle of beta testing RR-CKRTS Towers, Signals and BOD8's for detection. So far it is working quite well..

Don James.

atanisoft, 2019-10-23:
> 3. The differences between an ESP8266 and ESP32. The ESP32 has a built in can controller, the esp8226 does not

One other critical difference of the two, the esp32 is a dual core 240MHz SoC vs single core 80MHz SoC.

> How to setup the CDL screens.

The node itself provides an XML document to JMRI which generates the UI fields dynamically. The XML document describes the node's configurable attributes (name, description, IO, etc). As for using the CDI UI, that would be very node and task specific.

> How to add more i/o ports.

The [ESP32IOBoard](https://github.com/openmrn/OpenMRNLite/tree/master/examples/ESP32IOBoard) example defines 8 pins for input and 8 pins for output. There are a few other pins available on the ESP32 but there may be some restrictions on their usage. It is possible to use an I2C GPIO expander but there is currently no example for this or the support code for it (though I've seen it done with the OpenMRN stack).

> What and where is "print_all_packets()?

This method is defined [here](https://github.com/openmrn/OpenMRNLite/blob/master/src/openlcb/SimpleStack.hxx#L230-L235). It adds a DisplayPort instance to the stack which prints the CAN packets as they are sent/received. Something similar to this could be used for your CAN -> MQTT bridge but a different approach would be required for MQTT -> CAN.

> While I like LCC, it does have a few drawbacks.

What drawbacks have you found? I can agree with @balazsracz that there is a gap in the current offering of OpenMRNLite in that you can not easily specify the default behavior of the node when operating only on WiFi. This is something I'd like to improve upon after evaluating the full set of options available to us.