openmrn/OpenMRNLite #2: STM32 NUCLEO-144: compiling failed

AdrianBan, 2020-03-07:
Hi,

I'm trying to compile OpenMRNLite on STM32 Nucleo-144 and the compiler is complaining that is not finding netdb.h file. I've tried to find this file, but I don't know if is another library which is providing it.
Here is the output:

```
In file included from C:\Users\Adrian\Documents\Arduino\libraries\OpenMRNLite-0.1.1\src\freertos_drivers\esp32\Esp32WiFiManager.hxx:45,
                 from C:\Users\Adrian\Documents\Arduino\libraries\OpenMRNLite-0.1.1\src\freertos_drivers\esp32\Esp32WiFiManager.cpp:35:
C:\Users\Adrian\Documents\Arduino\libraries\OpenMRNLite-0.1.1\src/utils/SocketClient.hxx:44:10: fatal error: netdb.h: No such file or directory
   44 | #include <netdb.h>
      |          ^~~~~~~~~
compilation terminated.

exit status 1
Error compiling for board Nucleo-144.
```

Any ideas?

balazsracz, 2020-03-07:
If you are compiling for the STM32 nucleo, you should probably not be
including the ESP32 wifi driver (that's for a different chip). What is the
target / example application you were trying to compile?

AdrianBan, 2020-03-07:
Hi,

I was thought about that, but I didn't add any drivers for ESP32 WiFi.
Here is the code I'm trying to compile:

```
#include <cmsis_os.h>
#include <croutine.h>
#include <event_groups.h>
#include <FreeRTOS.h>
#include <FreeRTOSConfig.h>
#include <FreeRTOSConfig_Default.h>
#include <list.h>
#include <mpu_prototypes.h>
#include <portmacro.h>
#include <queue.h>
#include <semphr.h>
#include <stack_macros.h>
#include <STM32FreeRTOS.h>
#include <stream_buffer.h>
#include <task.h>
#include <timers.h>

#include <LwIP.h>
#include <OpenMRNLite.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```
Very basic compilation to check if the libs are ok.

Kind regards,
Adrian

balazsracz, 2020-03-07:
Aha, it seems that we have a .cpp file which is specific to the esp32 but
it is being compiled under all targets.

add an
#ifdef ESP32
...
#endif

around the entire contents of OpenMRNLite-0.1.1\src\freertos_drivers\
esp32\Esp32WiFiManager.cpp

We'll have to upstream that ifdef but this will get you going faster.

atanisoft, 2020-03-07:
I'll add that to my open PR for the same file...

AdrianBan, 2020-03-07:
Hi,

Thanks for the fix. Is not giving anymore this error, but I'm receiving this error:

```
In file included from C:\Users\Adrian\Documents\Arduino\libraries\OpenMRNLite-0.1.1\src/executor/Executor.hxx:50,
                 from C:\Users\Adrian\Documents\Arduino\libraries\OpenMRNLite-0.1.1\src/openlcb/SimpleStack.hxx:40,
                 from C:\Users\Adrian\Documents\Arduino\libraries\OpenMRNLite-0.1.1\src/OpenMRNLite.h:44,
                 from C:\Users\Adrian\Nextcloud\Electronic Projects\Arduino\DCCpp+LCC\Ethernet_Test\Ethernet_Test.ino:21:
C:\Users\Adrian\Documents\Arduino\libraries\OpenMRNLite-0.1.1\src/os/OSSelectWakeup.hxx: In member function 'int OSSelectWakeup::select(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, long long int)':
C:\Users\Adrian\Documents\Arduino\libraries\OpenMRNLite-0.1.1\src/os/OSSelectWakeup.hxx:231:16: error: 'ret' was not declared in this scope; did you mean 'creat'?
  231 |         return ret;
      |                ^~~
      |                creat
exit status 1
Error compiling for board Nucleo-144.
```

Kind regards,
Adrian

closed 2020-03-17 by balazsracz