openmrn/OpenMRNLite #3: STM32: compilation error: no SELECT methode

AdrianBan, 2020-03-09:
Hi,

After fixing #2 manually (until the PR will be pushed to the main source) I've tried to continue to compile the sources.
The source code:

```
#include <LwIP.h>
#include <OpenMRNLite.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

I'm receiving this error:

```
In file included from /home/adrian/Arduino/libraries/OpenMRNLite/src/executor/Executor.hxx:50,
                 from /home/adrian/Arduino/libraries/OpenMRNLite/src/openlcb/SimpleStack.hxx:40,
                 from /home/adrian/Arduino/libraries/OpenMRNLite/src/OpenMRNLite.h:44,
                 from /home/adrian/Nextcloud/Electronic-Projects/Arduino/DCCpp+LCC/DCC_LCC_basestation/DCC_LCC_basestation.ino:6:
/home/adrian/Arduino/libraries/OpenMRNLite/src/os/OSSelectWakeup.hxx:37:70: note: #pragma message: OSSelectWakeup: Starting including OSSelectWakeup
   37 | #pragma message ("OSSelectWakeup: Starting including OSSelectWakeup" )
      |                                                                      ^
/home/adrian/Arduino/libraries/OpenMRNLite/src/os/OSSelectWakeup.hxx:63:68: note: #pragma message: OSSelectWakeup: Ending including OSSelectWakeup
   63 | #pragma message ("OSSelectWakeup: Ending including OSSelectWakeup" )
      |                                                                    ^
/home/adrian/Arduino/libraries/OpenMRNLite/src/os/OSSelectWakeup.hxx: In member function 'int OSSelectWakeup::select(int, _types_fd_set*, _types_fd_set*, _types_fd_set*, long long int)':
/home/adrian/Arduino/libraries/OpenMRNLite/src/os/OSSelectWakeup.hxx:239:16: error: 'ret' was not declared in this scope; did you mean 'creat'?
  239 |         return ret;
      |                ^~~
      |                creat
exit status 1
Error compiling for board Nucleo-144.
```

The pragma messages I've inserted in the OSSelectWakeup.hxx:

```
#ifndef _OS_OSSELECTWAKEUP_HXX_
#define _OS_OSSELECTWAKEUP_HXX_

#pragma message ("OSSelectWakeup: Starting including OSSelectWakeup" )

#include <unistd.h>

#include "openmrn_features.h"
#include "utils/Atomic.hxx"
#include "os/os.h"

#if OPENMRN_FEATURE_DEVICE_SELECT
#pragma message ("OSSelectWakeup: Induding Devtab.hxx" )
#include "Devtab.hxx"
#endif

#if OPENMRN_HAVE_PSELECT
#pragma message ("OSSelectWakeup: Induding signal.h" )
#include <signal.h>
#endif

#ifdef __WINNT__
#pragma message ("OSSelectWakeup: Induding winsock2.h" )
#include <winsock2.h>
#elif OPENMRN_HAVE_SELECT
#pragma message ("OSSelectWakeup: Induding sys/select.h" )
#include <sys/select.h>
#endif

#pragma message ("OSSelectWakeup: Ending including OSSelectWakeup" )
```

And as you can see is actually not selecting any kind of SELECTOR. Going deeper in the code source I've notice those in openmrn_feature.h:

```
#ifdef __FreeRTOS__
/// Compiles the FreeRTOS event group based ::select() implementation.
#define OPENMRN_FEATURE_DEVICE_SELECT 1
/// Adds implementations for ::read ::write etc, with fd table.
#define OPENMRN_FEATURE_DEVTAB 1
/// Adds struct reent pointer to the FreeRTOS Task Priv structure and swaps it
/// in when the tasks are swapped in.
#define OPENMRN_FEATURE_REENT 1
#endif

/// @todo this should probably be a whitelist: __linux__ || __MACH__.
#if !defined(__FreeRTOS__) && !defined(__WINNT__) && !defined(ESP32) &&        \
    !defined(ARDUINO) && !defined(ESP_NONOS)
/// Uses ::pselect in the Executor for sleep and pkill for waking up.
#define OPENMRN_HAVE_PSELECT 1
#endif

#if defined(__WINNT__) || defined(ESP32) || defined(ESP_NONOS)
/// Uses ::select in the executor to sleep (unsure how wakeup is handled)
#define OPENMRN_HAVE_SELECT 1
#endif

#if defined(OPENMRN_HAVE_SELECT) || defined(OPENMRN_HAVE_PSELECT) || defined(OPENMRN_FEATURE_DEVICE_SELECT)
#define OPENMRN_FEATURE_EXECUTOR_SELECT
#endif

#if (defined(ARDUINO) && !defined(ESP32)) || defined(ESP_NONOS) ||             \
    defined(__EMSCRIPTEN__)
/// A loop() function is calling the executor in the single-threaded OS context.
#define OPENMRN_FEATURE_SINGLE_THREADED 1
#endif

```

I've noticed that there is no definition for __FreeRTOS__ in "STM32duino FreeRTOS" or in Arduino RTOS.
Second think I can't find any kind in the combination to match the selectors for OPENMRN. That's why, in my opinion, I can't compile at all on STM32.

Kind regards,
Adrian

atanisoft, 2020-03-09:
For the STM32 you may be better off using the full featured [OpenMRN](https://github.com/bakerstu/openmrn) library instead, this will entail moving away from the simpler Arduino APIs but it will likely be better in the long run.

balazsracz, 2020-03-09:
I have not tried compiling for the STM32 in the arduino environment in a
very long while. The support was never fully complete (in particular there
are dependencies ofthe current codebase on fd based file access APIs which
were never given on an Arduino -- the ESP32 is the only example where we
have it). We definitely do not have the FreeRTOS related parts of OpenMRN
ported to the Arduino compilation; this is why we call it OpenMRNLite.

I do a lot of compilation for STM32 natively under linux, but that's not a
good option under windows (lack of make and related utilities). Maybe the
linux subsystem would help but I have no experience.

In short what you are trying to do is not something someone before you did,
so there will be bumps on that road -- it needs work. If you are willing to
do that work, I will be happy to help.

Specifically, the select part needs to be disabled in STM32. When running
under the Arduino environmenmt we should be exercising the path that's
larked as SINGLE_THREADED. There is a bug in the code you quoted because it
does not have the #ifdefs correctly to work in the that path. That bug
needs to be fixed first to compile.
There will probably be more problems.

atanisoft, 2020-07-17:
@AdrianBan Can you please test with the latest update which adds the STM32 support to this library?

AdrianBan, 2020-07-17:
Yes, sure! I will give a try this evening.

AdrianBan, 2020-07-19:
Hi,

I've success to compile on NUCLEO-144 F767ZI:

```
Using library OpenMRNLite at version 1.0.2 in folder: C:\Users\Adrian\Documents\Arduino\libraries\OpenMRNLite 
Using library SrcWrapper at version 1.0.1 in folder: C:\Users\Adrian\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\libraries\SrcWrapper 
"C:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\xpack-arm-none-eabi-gcc\\9.2.1-1.1/bin/arm-none-eabi-size" -A "C:\\Users\\Adrian\\AppData\\Local\\Temp\\arduino_build_346191/Stm32CanSerial.ino.elf"
Sketch uses 44232 bytes (2%) of program storage space. Maximum is 2097152 bytes.
Global variables use 1792 bytes (0%) of dynamic memory, leaving 522496 bytes for local variables. Maximum is 524288 bytes.
```

On the other hand if I'm trying to compile on NUCLEO-64 F446RE F452RE is not working:

```
Generating function prototypes...
"C:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\xpack-arm-none-eabi-gcc\\9.2.1-1.1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb "@C:\\Users\\Adrian\\AppData\\Local\\Temp\\arduino_build_346191/sketch/build_opt.h" -c -Os -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit "-IC:\\Users\\Adrian\\Documents\\Arduino\\libraries\\OpenMRNLite\\examples\\Stm32CanSerial" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/avr" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/LL" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/usb" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/OpenAMP" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/usb/hid" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/usb/cdc" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/STM32F4xx_HAL_Driver/Inc" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/STM32F4xx_HAL_Driver/Src" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/STM32F4xx" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP/open-amp/lib/include" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP/libmetal/lib/include" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP/virtual_driver" -w -x c++ -E -CC -DSTM32F4xx -DARDUINO=10813 -DARDUINO_NUCLEO_F446RE -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"NUCLEO_F446RE\"" -DSTM32F446xx -DHAL_UART_MODULE_ENABLED "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\CMSIS\\5.5.1/CMSIS/Core/Include/" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/gcc/" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\CMSIS\\5.5.1/CMSIS/DSP/Include" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\variants\\NUCLEO_F446RE" "-IC:\\Users\\Adrian\\Documents\\Arduino\\libraries\\OpenMRNLite\\src" "-IC:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\libraries\\SrcWrapper\\src" "C:\\Users\\Adrian\\AppData\\Local\\Temp\\arduino_build_346191\\sketch\\Stm32CanSerial.ino.cpp" -o "C:\\Users\\Adrian\\AppData\\Local\\Temp\\arduino_build_346191\\preproc\\ctags_target_for_gcc_minus_e.cpp" -DARDUINO_LIB_DISCOVERY_PHASE
In file included from C:\Users\Adrian\Documents\Arduino\libraries\OpenMRNLite\src/OpenMRNLite.h:82,
                 from C:\Users\Adrian\Documents\Arduino\libraries\OpenMRNLite\examples\Stm32CanSerial\Stm32CanSerial.ino:37:
C:\Users\Adrian\Documents\Arduino\libraries\OpenMRNLite\src/freertos_drivers/stm32/Stm32Can.hxx:55:2: error: #error Dont know what STM32 chip you have.
   55 | #error Dont know what STM32 chip you have.
      |  ^~~~~
Using library OpenMRNLite at version 1.0.2 in folder: C:\Users\Adrian\Documents\Arduino\libraries\OpenMRNLite 
Using library SrcWrapper at version 1.0.1 in folder: C:\Users\Adrian\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\libraries\SrcWrapper 
exit status 1
Error compiling for board Nucleo-64.
``` 

Those 2 boards they have CAN bus also, so theoretically should compile the library on those chips.
But at least is compiling on Nucleo-144! :) I will give a try to test the library on my STM32.

atanisoft, 2020-07-19:
after [this](https://github.com/openmrn/OpenMRNLite/blob/master/src/freertos_drivers/stm32/Stm32Can.hxx#L53) line add:
```
#elif defined(STM32F446xx)
#include "stm32f4xx_hal_can.h"
#elif defined(STM32L452xx)
#include "stm32l4xx_hal_can.h"
```
That might take care of F446RE and L452RE (couldn't find an entry for M452RE in the stmduino boards.txt file).

AdrianBan, 2020-07-20:
Hi @atanisoft,

I've found that I need to do some additional definitions in Stm32can.cpp. I have a question regarding to this line:
```
#define CAN_CLOCK (cm3_cpu_clock_hz >> 2) // 54 MHz, sysclk/4
```
There is a formula where I can find how to calculate CAN Clock?
I want to add 2 more devices or maybe I will have few of them to add, if I have time.

atanisoft, 2020-07-20:
@balazsracz can you comment on the above? I'm not very familiar with STM32 unfortunatly.

AdrianBan, 2020-07-20:
Maybe I can get more information about CAN_CLOCK, where is used in the openMRN. I couldn't find it in the sources, only in that file.
Also I've notice that I need to change about 3 parts in total. I will create a patch and I will submit as PR, because most of the F4 and F7 MCU they support at least 1 CAN bus.

AdrianBan, 2020-07-20:
Success to compile it on F446RE, after adding the necessary code to the openMRN:

```
Using library OpenMRNLite at version 1.0.2 in folder: C:\Users\Adrian\Documents\Arduino\libraries\OpenMRNLite 
Using library SrcWrapper at version 1.0.1 in folder: C:\Users\Adrian\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\libraries\SrcWrapper 
"C:\\Users\\Adrian\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\xpack-arm-none-eabi-gcc\\9.2.1-1.1/bin/arm-none-eabi-size" -A "C:\\Users\\Adrian\\AppData\\Local\\Temp\\arduino_build_844764/Stm32CanSerialNode.ino.elf"
Sketch uses 113936 bytes (21%) of program storage space. Maximum is 524288 bytes.
Global variables use 2492 bytes (1%) of dynamic memory, leaving 128580 bytes for local variables. Maximum is 131072 bytes.
```

atanisoft, 2020-07-20:
Please submit the PR to https://github.com/bakerstu/openmrn

bakerstu, 2020-07-20:
Going purely from memory, (I think I was a the one that wrote the first iteration of the STM32 CAN driver), I believe the CAN peripheral clock frequency had a fixed divider based off of the main system clock on the devices we started with. This may or may not be true for other STM32 devices.

balazsracz, 2020-07-20:
The clock structure of the STM32 microcontrollers exhibits a baroque variety. Sometimes the CAN is on HCLK, sometimes on PCLK, sometimes there are two different PCLKs, and on some microcontrollers the HCLK and PCLK matches and sometimes the PCLK is divided from HCLK by an integer divisor. The divisor is often programmable too. There is a good reason the CAN middleware API asks for this value as an argument :)
You need to look at the programmer's reference manual that is specific to your microcontroller in order to see what the clock structure is and possibly at the code that sets up the oscillator block in the arduino core's startup routine to know how they initialize the divisors. Given that the arduino core is not tuned for low power consumption, usually the clocks are set to the maximum allowed, so maybe the reference manual is enough.

I would suggest to only look at MCUs that you have an example at hand. Making a guesswork of this is more error prone than worth. Testing becomes important.

I don't want you to waste too much time, so if you can't figure out what the right stanza is for the specific MCU that you have, let me know and I'll go look. You will have to test. But I will not go and do exhaustive research to get the values for every single MCU because that's not a good use of time and does not guarantee a good result either. I much rather recommend people to buy one of the few things that I picked as worth supporting (F091 F303 F767 and occasionally the F446).