openmrn/OpenMRNLite #5: mDNS not resolving when running ESP32WifiCanBridge

K-hastings, 2023-09-05:
Hello,

I am unable to get mDNS to resolve the _openlcb-can._tcp.local hostname. Other hosts running avahi-browse as well as the mDNS-SD_Extended example from Arduino-esp32 however resolve it no problem.

I created a newer library based on the instructions here: https://github.com/bakerstu/openmrn/blob/master/arduino/RELEASE.md
When trying with the 1.0.3 release package I am unable to compile due to Arduino-esp32 being 2.0.11.

Any suggestions to either get it to resolve or to be able to get in and manually connect it to the hub (either JMRI's or the one from OpenMRN are not being found automatically)

atanisoft, 2023-09-05:
> I am unable to get mDNS to resolve the _openlcb-can._tcp.local hostname.

It's a service name, not a hostname. Depending on the operating system you may have better luck with using the hostname "esp32mrn.local" instead.

> When trying with the 1.0.3 release package I am unable to compile due to Arduino-esp32 being 2.0.11.

This is a known issue and we are working on preparing an updated version of OpenMRNLite that works with the 2.x arduino-esp32 versions. There were major version changes from 1.x to 2.x and as a result the current OpenMRNLite versions are stuck on the 1.x versions (1.0.6 should work). If you have a Linux / MacOS env you can clone [https://github.com/bakerstu/openmrn](https://github.com/bakerstu/openmrn) and run `arduino/libify.sh` to export a version that should work with 2.x though.

K-hastings, 2023-09-05:
> > I am unable to get mDNS to resolve the _openlcb-can._tcp.local hostname.
> 
> It's a service name, not a hostname. Depending on the operating system you may have better luck with using the hostname "esp32mrn.local" instead.

esp32mrn refuses to resolve anywhere (with or without .local). No issues resolving _openlcb-can._tcp.local on everything except the ESP32 when running the example gateway. Even the ESP32 mDNS example can resolve it but not the can gateway example.
I haven't had any luck figuring out where it is trying to resolve to see if there is a code difference between the two examples.
> This is a known issue and we are working on preparing an updated version of OpenMRNLite that works with the 2.x arduino-esp32 versions. There were major version changes from 1.x to 2.x and as a [...]

Thats what I did when I said I created a newer library using the same link....

atanisoft, 2023-09-05:
> esp32mrn refuses to resolve anywhere (with or without .local).

Interesting, which OS and which tool are you using to resolve it?

> No issues resolving _openlcb-can._tcp.local on everything except the ESP32 when running the example gateway. 

That's interesting since it is not a hostname but is a service name (without the .local part).

> I haven't had any luck figuring out where it is trying to resolve to see if there is a code difference between the two examples.

Can you capture the serial console output from the gateway version? It should be starting up and searching for an existing LCC hub via mDNS (looking for _openlcb-can._tcp service name) and connecting to it when found. If you have a CAN interface on your PC you might be able to pull up the CDI from JMRI and check the configuration was seeded correctly on the esp32.

If you are re-using the esp32 between different examples I would strongly recommend clearing the flash memory between the examples to ensure it doesn't result in unexpected behaviors related to invalid persistent data.

K-hastings, 2023-09-05:
> Interesting, which OS and which tool are you using to resolve it?

Pop-os and ubuntu using `avahi-browse -d local _openlcb-can._tcp --resolve -t` correctly gets the txt and ip records.
Using the mDNS-SD_extended example from Arduino-esp32 also grabs the IP of the hub from _openlcb-can._tcp.
As soon as I use the ESP32WifiCanGateway it and it alone seems to never be able to find and get any information from _openlcb-can._tcp. Just keeps looping that it failed.

> Can you capture the serial console output from the gateway version? It should be starting up and searching for an existing 
> LCC hub via mDNS (looking for _openlcb-can._tcp service name) and connecting to it when found. 

The serial console is how I know it is failing. 

> If you have a CAN interface on your PC you might be able to pull up the CDI from JMRI and check the configuration was 
> seeded correctly on the esp32.

I don't have a CAN interface for my computer. This is what I was hoping the ESP32 would be.
[image]

K-hastings, 2023-09-05:
Looks like something was stuck in the EEPROM despite clearing it previously. It is working now.

closed 2023-09-05 by K-hastings

atanisoft, 2023-09-05:
Glad you were able to sort out what was causing it. From the console output that log message for Uplink is indicating that it did not find any other node(s) advertising the service name (JMRI doesn't advertise it today that I know of).

K-hastings, 2023-09-05:
It does when you launch the hub. Though for most of my testing I had been using the hub app included with openmrn.

This e-mail and any attachments may contain information that is confidential to Ross Video.

If you are not the intended recipient, please notify me immediately by replying to this message. Please also delete all copies. Thank you.

atanisoft, 2023-09-05:
If you are using the openmrn hub make sure to pass "-m" so it does advertise itself, the esp32 should be able to locate it and connect automatically.

K-hastings, 2023-09-05:
I have that flag. It is advertising. Has been the whole time. Thats why I could see it with avahi-browse. There was an issue with the contents of the eeprom on the esp near I could tell. Hence why this issue has been closed and resolved.

balazsracz, 2023-09-06:
Could you capture the same log as what the screenshot above has?
I'd like to see the example when it is working.