openmrn/OpenMRNLite #6: ESP32IOBoard Example Compile Error

whoarreyou, 2024-06-08:
Hi

I tried to compile the ESP32IOBoard example from the Arduino V2.0.0 library but I am getting the errors shown below.
Sorry if I am just doing something wrong.

In file included from C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src/freertos_drivers/arduino/PWM.hxx:37,
                 from C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src\freertos_drivers\esp32\Esp32Ledc.hxx:38,
                 from C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src\freertos_drivers\esp32\Esp32Ledc.cpp:42:
C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src\freertos_drivers\esp32\Esp32Ledc.hxx: In member function 'void openmrn_arduino::Esp32Ledc::fade_channel_over_time(unsigned int, uint32_t, uint32_t, ledc_fade_mode_t)':
C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src\freertos_drivers\esp32\Esp32Ledc.hxx:191:22: error: 'pthread_once' was not declared in this scope; did you mean 'pthread_once_t'?
  191 |         HASSERT(0 == pthread_once(&ledcFadeOnce_, &Esp32Ledc::ledc_fade_setup));
      |                      ^~~~~~~~~~~~
C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src/utils/macros.h:97:31: note: in definition of macro 'HASSERT'
   97 | #define HASSERT(x) do { if (!(x)) { printf("Assertion failed in file " __FILE__ " line %d: assert(%s)\n", __LINE__, #x); g_death_file = __FILE__; g_death_lineno = __LINE__; assert(0); abort();} } while(0)
      |                               ^
C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src\freertos_drivers\esp32\Esp32Ledc.cpp: At global scope:
C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src\freertos_drivers\esp32\Esp32Ledc.cpp:47:43: error: 'PTHREAD_ONCE_INIT' was not declared in this scope; did you mean 'OS_THREAD_ONCE_INIT'?
   47 | pthread_once_t Esp32Ledc::ledcFadeOnce_ = PTHREAD_ONCE_INIT;
      |                                           ^~~~~~~~~~~~~~~~~
      |                                           OS_THREAD_ONCE_INIT
In file included from C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src/os/OS.hxx:40,
                 from C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src/executor/Notifiable.hxx:39,
                 from C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src\freertos_drivers\esp32\Esp32HardwareTwai.cpp:75:
C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src\freertos_drivers\esp32\Esp32HardwareTwai.cpp: In member function 'void openmrn_arduino::Esp32HardwareTwai::hw_init()':
C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src\freertos_drivers\esp32\Esp32HardwareTwai.cpp:1008:26: error: too few arguments to function 'bool twai_hal_init(twai_hal_context_t*, const twai_hal_config_t*)'
 1008 |     HASSERT(twai_hal_init(&twai.context));
      |             ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src/utils/macros.h:97:31: note: in definition of macro 'HASSERT'
   97 | #define HASSERT(x) do { if (!(x)) { printf("Assertion failed in file " __FILE__ " line %d: assert(%s)\n", __LINE__, #x); g_death_file = __FILE__; g_death_lineno = __LINE__; assert(0); abort();} } while(0)
      |                               ^
In file included from C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src\freertos_drivers\esp32\Esp32HardwareTwai.cpp:69:
C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32/include/hal/include/hal/twai_hal.h:87:6: note: declared here
   87 | bool twai_hal_init(twai_hal_context_t *hal_ctx, const twai_hal_config_t *config);
      |      ^~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

atanisoft, 2024-06-08:
Add this line:
```
#include <pthread.h>`
```

After [this](https://github.com/openmrn/OpenMRNLite/blob/f45ea6e497f2c6595e393bef785fc7b8a3496270/src/freertos_drivers/esp32/Esp32Ledc.hxx#L43) line and it should fix it.

We'll get a fix synced in for this.

whoarreyou, 2024-06-08:
Yes, great, that has fixed the first error. I still have the below error in Esp32HardwareTwai.cpp

In file included from C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src/os/OS.hxx:40,
                 from C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src/executor/Notifiable.hxx:39,
                 from C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src\freertos_drivers\esp32\Esp32HardwareTwai.cpp:75:
C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src\freertos_drivers\esp32\Esp32HardwareTwai.cpp: In member function 'void openmrn_arduino::Esp32HardwareTwai::hw_init()':
C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src\freertos_drivers\esp32\Esp32HardwareTwai.cpp:1008:26: error: too few arguments to function 'bool twai_hal_init(twai_hal_context_t*, const twai_hal_config_t*)'
 1008 |     HASSERT(twai_hal_init(&twai.context));
      |             ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src/utils/macros.h:97:31: note: in definition of macro 'HASSERT'
   97 | #define HASSERT(x) do { if (!(x)) { printf("Assertion failed in file " __FILE__ " line %d: assert(%s)\n", __LINE__, #x); g_death_file = __FILE__; g_death_lineno = __LINE__; assert(0); abort();} } while(0)
      |                               ^
In file included from C:\Users\User\Documents\Arduino\libraries\OpenMRNLite\src\freertos_drivers\esp32\Esp32HardwareTwai.cpp:69:
C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32/include/hal/include/hal/twai_hal.h:87:6: note: declared here
   87 | bool twai_hal_init(twai_hal_context_t *hal_ctx, const twai_hal_config_t *config);
      |      ^~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

atanisoft, 2024-06-09:
You are running with Arduino-3.0.x which is not currently supported by the OpenMRNLite code, there should be an updated version in the near future to support 3.0.x versions.

For now I'd suggest downgrade to Arduino-esp32 2.0.17 (latest in the 2.0.x line). Keep the update for `pthread.h` and you shouldn't face any issues.

whoarreyou, 2024-06-10:
I did as you suggested and downgraded. It now compiles and works great. Thanks for your help.

atanisoft, 2025-08-15:
This should be fixed in the latest release which offers support for the newer arduino-esp32 3.3.x versions.

closed 2025-08-15 by atanisoft