openmrn/OpenMRNLite #7: example Stm32CanUSB: compilation error

nopxor, 2024-09-04:
Hello,
I try to use OpenMRNLite (version 2.0.0) with the example Stm32CanUSB
I use a NUCLEO 64 board: F103RB
I use Arduino IDE version 1.8.19 with STM32 MCU based boards version 2.8.1
I set TOOLS -> USB support: "CDC (generic 'Serial' supersede U(S)ART)"

First of all, I have got some errors of compilation in files:
src\os\MDNS.hxx
src\os\MDNS.cpp
src\utils\Base64.cpp
src\utils\format_utils.hxx
src\utils\format_utils.cpp
I solve the problem by adding #include <stdint.h> in this files

Now at linking, I have 2 errors of multiple definition, 
maybe it comes from the #define line 55 and 57 of the file:
src\freertos_drivers\stm32\Stm32Can.cpp

Here is my "debug window":
`Linking everything together...
"C:\\Users\\Eric\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\tools\\xpack-arm-none-eabi-gcc\\13.2.1-1.1/bin/arm-none-eabi-gcc" -mcpu=cortex-m3 -mthumb -Os -DNDEBUG --specs=nano.specs -Wl,--defsym=LD_FLASH_OFFSET=0x0 -Wl,--defsym=LD_MAX_SIZE=131072 -Wl,--defsym=LD_MAX_DATA_SIZE=20480 -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common "-Wl,--default-script=C:\\Users\\Eric\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.8.1\\variants\\STM32F1xx\\F103R(8-B)T/ldscript.ld" "-Wl,--script=C:\\Users\\Eric\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.8.1\\system/ldscript.ld" "-Wl,-Map,C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096/Stm32CanUSB.ino.map" -Wl,--no-warn-rwx-segments -o "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096/Stm32CanUSB.ino.elf" "-LC:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096" -Wl,--start-group "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\sketch\\Stm32CanUSB.ino.cpp.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\sketch\\requiredLibraries.cpp.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\OpenMRNLite\\OpenMRNLite.a" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\syscalls.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HardwareTimer.cpp.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\new.cpp.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_adc.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_adc_ex.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_can.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_cec.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_comp.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_comp_ex.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_cordic.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_cortex.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_crc.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_crc_ex.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_cryp.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_cryp_ex.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_dac.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_dac_ex.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_dcache.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_dcmi.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_dcmi_ex.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_dfsdm.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_dfsdm_ex.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_dma.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_dma2d.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_dma_ex.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_dsi.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_dts.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_eth.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_eth_ex.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_exti.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_fdcan.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_firewall.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_flash.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_flash_ex.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_flash_ramfunc.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_fmac.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_fmpi2c.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_fmpi2c_ex.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_fmpsmbus.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_fmpsmbus_ex.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_gfxmmu.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_gfxtim.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_gpio.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_gpio_ex.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_gpu2d.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_gtzc.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_hash.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_hash_ex.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_hcd.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_hrtim.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_hsem.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_i2c.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_i2c_ex.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_i2s.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper\\HAL\\stm32yyxx_hal_i2s_ex.c.o" "C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_902096\\libraries\\SrcWrapper [... cut ...]
Utilisation de la bibliothèque SrcWrapper version 1.0.1 dans le dossier: C:\Users\Eric\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.8.1\libraries\SrcWrapper 
Utilisation de la bibliothèque USBDevice version 1.0.0 dans le dossier: C:\Users\Eric\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.8.1\libraries\USBDevice 
exit status 1
Erreur de compilation pour la carte Nucleo-64`

I tried to use another NUCLEO 64 board: F401RE, but it seems not recognized by the file:
src\freertos_drivers\stm32\Stm32Can.cpp

balazsracz, 2024-09-04:
Hi,

Unfortunately the F103 does not support using both USB and CAN-bus
peripheral at the same time. The two peripherals have some shared resource
for hardware buffering, and you can either use CAN-bus or you can use USB.
I recommend the F303RE for using both usb and CAN-bus at the same time. I
also have been using the F091RC a lot and know that that one works solid.

Balazs

balazsracz, 2024-09-04:
On the Nucleo board you also have a "usb" via the debug port, and that
looks like a Serial to us. So you can use the CanSerial example and turn
off USB support in the tools menu entirely.

nopxor, 2024-09-04:
Thanks a lot for your response.
I will try with the recommended F303RE or F091RC.

closed 2024-09-17 by balazsracz