openmrn/OpenMRNLite #8: ESP32IOBOARD no output

nopxor, 2024-09-15:
I have managed a test CAN network with a ESP32SerialBridge and 2 ESP32IOBoard.
I have connected some pushbuttons with pullup resistors as inputs and some leds with resistors as outputs. 
In JMRI, I can see the pushbuttons events but when I associate the events with the outputs, the leds do not light.

atanisoft, 2024-09-16:
Can you check the console output from the esp32? There usually are clues there as to what went wrong.

nopxor, 2024-09-16:
Thank you for the response.
The console output seems fine.
I did another test and the outputs worked fine.
But a few moments later, I lost the outputs again
certainly a dirty wild wiring ; )
This partial operation gives me good hope.

closed 2024-09-17 by balazsracz